d3ngit/djihdfpv_mavlink_to_msp_V2 #3: Arming not detected

FranzAr, 2019-12-20:
I installed the arduino on my plane and all seems to work fine besides arming is not detected by the goggles anymore.

d3ngit, 2019-12-21:
Ok. I managed to test this. For me it works as expected, arming starst recording, disarming stops it. It detects arming as it should, Inav 2.3.0, configurator 2.3.2. Maybe try to reupload the fw or redo the settings.

https://static.rcgroups.net/forums/a...1-177-iNav.png

Another thing. The first time I tried I only enabled Arming in the modes tab and it didn't work, but I also haven't had the battery connected to the FC. The second try connected the battery and added the Angle, Horizon,Manual modes to one switch. So I don't know if the enabling of modes is related somehow.

FranzAr, 2019-12-21:
Thanks for your help, I really appreciate that.

I tried a million things now and I can't get it to work. Tried your diff, tried the hex from github, thousand combinations of flight modes etc. All configured OSD items appear on the screen and work like expected. Just this f*** arming detection doesn't. It worked when I had a direct connection from the FC to the air unit. I have no further ideas, so sad. I can use it like it is now, start recording manually, disable auto power control. But it'd definitely be nice to have the auto arming detection.

Does it work for everybody else?

naitse, 2019-12-22:
My arming is also not being detected I have to manually start the recording using the goggles button every time

d3ngit, 2019-12-22:
Me and another user confirmed it's working fine. I tried at least 30 times consequently and worked every single time.

Could you give some more information:
-Do you have an SD car in air unit? A good card, not full, etc...
-Did you try to reset the goggles settings?
-Did you connect everything like in the arduino photo, only TX for airunit? Also grounds? I'm using the extra ground wire from the airunit for the serial port.
-Does the arming work fine when hooked directly to the inav FC, or another FC with betaflight?
-If you set the recording to start only in the goggles and not airunit? Does it make any difference?

If you find a solution and fix the code yourself please share. Thanks.

FranzAr, 2019-12-22:
I tried everything you mentioned and ist doesn't make any difference.

naitse, 2019-12-23:
**only for INAV** please reduce the mavlink telemetry speed to 38400 or 19200, in this way the HEARTBEAT msg gets received by the arduino and arming starts to work

FranzAr, 2019-12-23:
Tried at 19200 and it's working. Yeah, great, love it

d3ngit, 2019-12-23:
Thanks. Good to know.

d3ngit, 2019-12-25:
I just updated the code with an improved softserial library. Try now with normal speeds and report if it's working. Thanks.

FranzAr, 2019-12-25:
Did a quick test and it arming is detected at 57600 :-)

closed 2019-12-28 by d3ngit