d3ngit/djihdfpv_mavlink_to_msp_V2 #5: Compass working?

spyder0069, 2019-12-28:
Just started working with this tonight using an ESP32 chip. I am getting voltage/current/flightmodes and such but I do not get the compass, distance to home, home direction to show.  I do show satellites and lat/lon.  The sat icon is blinking.

Do these other options show up once armed or are they supposed to show up right away?

This is using a pixhawk if it matters.  I appreciate the work and any help.  :^)

d3ngit, 2019-12-28:
If sats are blinking you don't have gps lock. Also check if your mavlink rates are like this in your parameters, and use the latest ino because there have been updates: 

(this is for SR1, serial port 1)
SR1_ADSB,5
SR1_EXT_STAT,2
SR1_EXTRA1,5
SR1_EXTRA2,2
SR1_EXTRA3,2
SR1_PARAMS,10
SR1_POSITION,2
SR1_RAW_CTRL,1
SR1_RAW_SENS,2
SR1_RC_CHAN,5

spyder0069, 2019-12-28:
I have the SR mavlink rates set properly.  I just pulled the code last night.  I have 13 satellites found and my GPSSTATUS=3 which means I have 3d fix.  My GPSHDOP is 0.79.  

I am indoors so I have not armed.  Does it require arming to see the other items?  I would have though things like the compass would show without arming.

I have this all working on a minimosd so the output should all be there.  This is also running a crossfire with control over mavlink on telemetry 1.

d3ngit, 2019-12-28:
No, you don't need to arm. It should work without.

spyder0069, 2019-12-28:
I found the problem.  Your code needed to be changed to: `if(fix_type > 2 && set_home == 1 && gps_lat != 0 && gps_lon != 0 && numSat > 5){`
 
This is because my longitude which:  gps_lon=-892446495

Now I have a solid satellite icon.  It sets the _home variables.

I still do not get the compass bar or other home icons.  Here is my config:

`const uint16_t osd_rssi_value_pos = 2048;
const uint16_t osd_main_batt_voltage_pos = 2304;
const uint16_t osd_crosshairs_pos = 234;
const uint16_t osd_artificial_horizon_pos = 234;
const uint16_t osd_horizon_sidebars_pos = 234;
const uint16_t osd_item_timer_1_pos = 234;
const uint16_t osd_item_timer_2_pos = 234;
const uint16_t osd_flymode_pos = 234;
const uint16_t osd_craft_name_pos = 234;
const uint16_t osd_throttle_pos_pos = 234;
const uint16_t osd_vtx_channel_pos = 234;
const uint16_t osd_current_draw_pos = 2368;
const uint16_t osd_mah_drawn_pos = 2336;
const uint16_t osd_gps_speed_pos = 2273;
uint16_t osd_gps_sats_pos = 2080;
const uint16_t osd_altitude_pos = 2295;
const uint16_t osd_roll_pids_pos = 234;
const uint16_t osd_pitch_pids_pos = 234;
const uint16_t osd_yaw_pids_pos = 234;
const uint16_t osd_power_pos = 234;
const uint16_t osd_pidrate_profile_pos = 234;
const uint16_t osd_warnings_pos = 2545;
const uint16_t osd_avg_cell_voltage_pos = 2072;
const uint16_t osd_gps_lon_pos = 2432;
const uint16_t osd_gps_lat_pos = 2400;
const uint16_t osd_debug_pos = 234;
const uint16_t osd_pitch_angle_pos = 234;
const uint16_t osd_roll_angle_pos = 234;
const uint16_t osd_main_batt_usage_pos = 234;
const uint16_t osd_disarmed_pos = 234;
const uint16_t osd_home_dir_pos = 2113;
const uint16_t osd_home_dist_pos = 2144;
const uint16_t osd_numerical_heading_pos = 234;
const uint16_t osd_numerical_vario_pos = 234;
const uint16_t osd_compass_bar_pos = 2095;
const uint16_t osd_esc_tmp_pos = 234;
const uint16_t osd_esc_rpm_pos = 234;
const uint16_t osd_remaining_time_estimate_pos = 234;
const uint16_t osd_rtc_datetime_pos = 234;
const uint16_t osd_adjustment_range_pos = 234;
const uint16_t osd_core_temperature_pos = 234;
const uint16_t osd_anti_gravity_pos = 234;
const uint16_t osd_g_force_pos = 234;
const uint16_t osd_motor_diag_pos = 234;
const uint16_t osd_log_status_pos = 234;
const uint16_t osd_flip_arrow_pos = 234;
const uint16_t osd_link_quality_pos = 234;
const uint16_t osd_flight_dist_pos = 2176;
const uint16_t osd_stick_overlay_left_pos = 234;
const uint16_t osd_stick_overlay_right_pos = 234;
const uint16_t osd_display_name_pos = 234;
const uint16_t osd_esc_rpm_freq_pos = 234;
const uint16_t osd_rate_profile_name_pos = 234;
const uint16_t osd_pid_profile_name_pos = 234;
const uint16_t osd_profile_name_pos = 234;
const uint16_t osd_rssi_dbm_value_pos = 234;
const uint16_t osd_rc_channels_pos = 234;`

d3ngit, 2019-12-28:
Wait. Are we sure you're not trying to set the osd elements for compass etc. that are set to 234 by default? This is a hack and will display the two numerical direction and dist to home in the lower right corner in place of pitch and roll angles like in the pic.

d3ngit, 2019-12-28:
Thanks. I'll fix the gps check.

spyder0069, 2019-12-28:
Ah.  So the compass bar and other items in the betaflight config are not fully implemented.  I got you.

spyder0069, 2019-12-28:
Was there a list of items that are working?

d3ngit, 2019-12-28:
Yes. Maybe in the future I'll add them but I'll try to fix the current bugs.

d3ngit, 2019-12-28:
The ones that are set to more that 234 are working, and main batt voltage. Others need to be sent too so they are all there.

spyder0069, 2019-12-28:
Sounds good.  Thanks for your input and efforts.  I can probably pull things off the breadboard now and try a flight.  The ESP32 worked pretty well for this as you have 3 hardware serial ports to play with once you relocate serial1 to other pins as to not interfere with the flash memory.  Leaves the serial0 for programming/debugging off the usb on the dev boards.  :^)

d3ngit, 2019-12-28:
Ok. Good luck!

closed 2019-12-28 by d3ngit