d3ngit/djihdfpv_mavlink_to_msp_V2 #6: RSSI question

spyder0069, 2019-12-28:
When I start up without the tx on your rssi value is at zero but the osd shows 99.  

Once I power up my crossfire the rssi value goes to 1023.

When I power down the crossfire I get a constant value of 20 and the osd shows 1.

Any input?

d3ngit, 2019-12-28:
Arupilot sends the rssi number over mavlink in the range of 0-255 and it gets scaled up to 0-1023 in the code to be compatible. I don't know exactly but I have mine set to channel 8 pwm and it auto pulls it from there and displays it in the rssi field.

spyder0069, 2019-12-28:
I have the same setup.  I assume you can reproduce the 99 rssi on powerup with no transmitter on?

My rxrssi in missionplanner shows 0 and again your code shoes zero rssi but it still showed 99 on the osd.  Guessing something with the scaling math going beyond zero or something?

With my crossfire and looking at the channel output with transmitter off (first power on) it doesn't output anything (zero pwm).  Once powered on CH8 goes to 2000.  Then once powered down and it goes into failsafe CH8 goes to 1020.  That is why its showing the 1%.  So that sounds like it would need to be a crossfire fix.

spyder0069, 2019-12-28:
Maybe your receiver is outputting on CH8 a valid pwm on boot without transmitter powered up so you never saw the 99 issue.  Check your status in planner and see if you are getting ch8 pwm when you first power on.

d3ngit, 2019-12-28:
99% is normal for 0 on boot, also shows 14 sats when they're 0. This is a dji thing. Do you get the rssi in between or not. I didn't get it.

d3ngit, 2019-12-28:
This could be a really easy fix if that bothers you, just put 1 instead of 0 as the default rssi variable. uint16_t rssi = 0;

spyder0069, 2019-12-28:
Still had the 1% after that change.  Its not a big deal just didn't know if anyone had mentioned it and figured I would give some feedback.  :^)

spyder0069, 2019-12-28:
You can close this if that is expected behavior.   Thanks!

d3ngit, 2019-12-28:
Ok. It's not a problem if it works as expected when you fly. On boot everything can happen with default values.

closed 2019-12-28 by d3ngit