d3ngit/djihdfpv_mavlink_to_msp_V2 #7: Map Throttle Position to unused element

FranzAr, 2020-01-05:
Hi, I'd really like to see my throttle position in the OSD.
Would it be possible to map it to an unused element.
I played around a bit and found out that DISARMED, FLYMODE (already toggling with Alt/Speed), MAIN BATTERY VOLTAGE are working  and not really used.
Greets
Franz

d3ngit, 2020-01-05:
This would be pretty easy to do. I'll look at it today. what about the main_batt_usage, looks like a slider, or the roll/pitch pids but in the range of 0%-100%. You need it in percents or PWM values?

FranzAr, 2020-01-05:
Percentage would be great. Element of your choice ;-) Thanks mate!

d3ngit, 2020-01-05:
I have it already working with the ROLL pid element and it's working fine. I'll just do a few more tests in the simulator and update the code soon.

d3ngit, 2020-01-05:
Done.

closed 2020-01-05 by d3ngit

FranzAr, 2020-01-05:
Thanks mate! I'll check that out tonight :-)

FranzAr, 2020-01-05:
Can confirm, works like a charm. Thanks