d3ngit/djihdfpv_mavlink_to_msp_V2 #8: unable to flash Nano_INAV_Plane.hex

watsonVon, 2020-01-10:
I am using the following command (that used to work with the previous .HEXes):

_/Users/andorn/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/andorn/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega328p -carduino -P/dev/cu.usbserial-14530 -b57600 -D -Uflash:w:Nano_INAV_Plane.hex:i_

With the result:

_avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file "Nano_INAV_Plane.hex"
avrdude: invalid record at line 671 of "Nano_INAV_Plane.hex"
avrdude: read from file 'Nano_INAV_Plane.hex' failed

avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you._

**BTW: Thank you so much for this project, great work**

d3ngit, 2020-01-10:
Hmmm. I need to look at it. Can you try with XLoader? I usually use that for hex files, but it's possible something went wrong in the compilation process.

watsonVon, 2020-01-10:
After finding a computer with windows, i got just a "upload failed" from XLoader. I have not found any log, so that is not very helpful i guess.  

Maybe someone could confirm that the hex is working or not ?

Thank you for your help.

d3ngit, 2020-01-10:
Yes. I just checked and the problem is real. Happens even if I try to upload it directly from the arduino ide. Interesting. I have to investigate this a little bit. In the meantime you can still use the Arduplane hex, eveything is the same, just your flight mode names will be named in the ardupilot way.

watsonVon, 2020-01-10:
_avrdude: reading input file "Nano_Arduplane.hex"
avrdude: invalid record at line 671 of "Nano_Arduplane.hex"
avrdude: read from file 'Nano_Arduplane.hex' failed_

¯\_(ツ)_/¯

d3ngit, 2020-01-11:
Ok. The Arduino nano I have on my plane that I have flashed (too)many times seems to be damaged somehow and I can't upload any hex file anymore on it. I'll pull it out and try reflashing the bootloader. But I tried with another spare one and all the hex files seem to be uploading and working fine. So it's nothing wrong with the hex files. Your problem must be somewhere else. You should try to download the arduino ide and try uploding some blink sketches that are included to see if evreything uploads fine and then try to upload the osd code(you have to copy the contents of the github Arduino_libraries folder to your local /libraries/ folder where arduino ide is installed).

d3ngit, 2020-01-11:
I can confirm once again the hexes work fine. My arduino nano corrupted the bootloader for some reason the first time I tested. Now I reflashed the bootloader with an AVR programmer and everything is working fine again and it flashes all hex files from usb and they work as expected.

closed 2020-01-11 by d3ngit