d3ngit/djihdfpv_mavlink_to_msp_V2 #9: error durin compilation

svacinkaj, 2020-01-27:
Please help me. I am just newbie in programing. I can't upload the code to the board.

[image]

d3ngit, 2020-01-27:
Copy the contents fo the Arduino_libraries fodler into your local Arduino /libraries/ folder location.

svacinkaj, 2020-01-27:
Thanks :-)
and next trouble
[image]

d3ngit, 2020-01-28:
What plane type are you compiling for? Which version of the arduino ide? I think it still compiled ok at the end and that are just warnings. Did you try to upload it?

iskymaster, 2020-05-02:
Hi,

where do i copy these files to my local computer?

#include "MSP_OSD.h"
#include "flt_modes.h"
#include "OSD_positions_config.h"

I am getting an error message.

ramiss, 2020-07-31:
By the way, I just removed these two packed attributes in the header file and it complies and runs fine with zero warnings.