d3ngit/djihdfpv_mavlink_to_msp_V2 #13: Invalid initialisation of reference

ProRow, 2020-10-24:
Hi there,

I'm really new to this so please bare with me!!

I've followed the instructions at [https://github.com/ukmook/qUark/tree/master/Mav-To-MSP#quark-flight-electronics] and no matter what I do am getting the following error which is breaking the build:

djihdfpv_mavlink_to_msp_V2:27:31: error: invalid initialization of reference of type 'HardwareSerial&' from expression of type 'Serial_'
   HardwareSerial &mspSerial = Serial;
                               ^~~~~~
exit status 1
invalid initialization of reference of type 'HardwareSerial&' from expression of type 'Serial_'

I have the 3 libraries in my libraries folder: MSP, AltSoftSerial and GCSMAVLink.
Running Arduino 1.8.13 On WinOS

I'm also getting the MSP Servo configs error previously mentioned in another post but happy these can be ignored.

Can anyone point me in the direction towards solving the problem?

Thanks in advance for any support.

[image]

d3ngit, 2020-10-25:
Did you select Arduino Nano Atmega 328P from the boards?

For me it compiles fine unless I select the wrong microcontroller.

ProRow, 2020-10-25:
That worked - thanks for the help and apologies for taking your time.

Considering my success here, can't wait to get into the wiring!!

closed 2020-10-25 by ProRow